EPFLRocketTeam/SupersonicTestRocket #12: Make the serial console printing work with the new packets

joshdboss, 2022-01-18:
The new packet format is more versatile, although not yet compatible with the serial console printing.

spacewolfXfr, 2022-02-09:
It should be OK for the sensors with wrappers. Testing is needed, as always.
 Some possible improvements:
- Add colors (define the color scheme to use)
- Add units (they are not always documented)

More general issue/question : `dataAcquisition` always does the setup and cleanup phase, but is called several time (at least once for `setup` in `main.cpp`, then in `loop`). Is this something which should be fixed ? (Split the function in `dataAcquisition_setup`, `..._core` and `..._cleanup` ?)

spacewolfXfr, 2022-03-22:
Done here #50

closed 2022-03-22 by spacewolfXfr