EPFLRocketTeam/SupersonicTestRocket #3: Document the c++ code using Doxygen

joshdboss, 2022-01-18:
The code is not currently documented. This should be done to allow for continuity or for onboarding of new members.

spacewolfXfr, 2022-02-04:
[Link to download the Doxygen tool](https://www.doxygen.nl/download.html) (Generate documentation from Doxygen-like commented code; helper to produce such comments are available as plugins in VS Code)

spacewolfXfr, 2022-02-04:
My Doxygen setup is ready! Now the code itself has to be commented [with Doxygen style](https://www.doxygen.nl/manual/docblocks.html) (the whole manual is also useful for detailed points, such as formulas)

spacewolfXfr, 2022-02-07:
[Doxygen displayed in Github](https://github.com/satu0king/Github-Documentation-With-Doxygen)

spacewolfXfr, 2022-02-07:
[Customize Doxygen](https://www.doxygen.nl/manual/customize.html#xmlgenerator)
(Bottom of the page : interface XML output with Sphinx for Python)

joshdboss, 2022-02-13:
Would it be possible to setup a GitHub Pages or something similar so that the documentation is visible directly online without having to clone the code locally? From what I read it should be possible and would make things more accessible for team members that don't necessarily need/want to clone the repository.